hugen79/NanoVNA-H #1: Proposal: add suffix to your product name (eg. NanoVNA-H)

edy555, 2019-08-13:
Hi hugen, 

I'm an original developer of NanoVNA.
When I knew that you made the clone of my project, I very surprised that you made PCB from schematics, and as an enthusiast, I felt pleased with your challenges such as frequency range expansion and your original pc software.

However, I was annoyed that you sell your clone without any prior notice to me. Though I had a plan to forge and sell my product, it becomes difficult.
Furthermore, your act cause that your design material was stolen by other clone makers, and quite many units were sold in aliex, ebay and also amazon. Those include worse clone as you say. This is a worrying state of affairs I think, and you might agree.

To distinguish any unexpected clones, I propose that you should change your product name by adding a suffix such as NanoVNA-H from your name, and it should be shown in the market and printed on your product.
And also I'd like to publish the name of qualified products, so customers become able to avoid worse clone.

I hope you think about this issue seriously.
Regards,

hugen79, 2019-08-16:
Dear Edy，

I am so glad to hear from you. First thanks for your approval and say sorry to make you annoyed.  

At the beginging I made this NanoVNA just because I needed a portable device to measure filter and antenna. I tried to make PCB and achieved very satisfactory results. The performance of nanoVNA is very exciting, and your software is very excellent too. So I shared the production experience in 38hot. Some HAM contacted me to buy nanoVNA.  I made 50 sets of hardware for sale to HAM by hand all of the parts, and provided the schematic diagram and source code. After testing, they thought the nanoVNA was excellent and then I got more orders as I expected. At the same time, I found bad clones on eday. And their photos showed mixer problems on their machines. Apparently they didn't find the problems.  And some traders contacted me to buy NanoVNA for sale on eBay. I realized that it was no long a simple DIY and sharing.  It was being commercialized.  I tried to contact you through Twitter, but maybe because of my poor English expression, I didn't get a response. Soon, more and more clones appreared on the market. And seems someone cloned this NanoVNA misled ebay's sellers to promote their products. Things become out of my control as an enthusiast. 

Maybe because I have stated your design concept every sites, maybe because I shared my experience and provide long term help as I can, or maybe because my products still have some advantages,  and thanks to the support of all HAMs, I can still get some more orders now.  And thank you most for not blaming me but giving me advice. I have added H in the name as NanoVNA-H and added a copyright declaration when open the VNA.  Hope your design will benefits more enthusiasts.

THANK YOU  & SORRY. 

Sincerely,
Hugen 

PS: Bec of my poor English, pls forgive me for my slowly response. I need sometime to translate and ask for help from my friends.

gabriel-tenma-white, 2019-09-29:
I'm pretty sure if you write Japanese hugen can understand it ;)

closed 2019-09-29 by edy555