hugen79/NanoVNA-H #11: Cross compiling on a Raspberry Pi...

dl9sec, 2019-11-05:
Hi,

i tried to cross compile the stuff on a Raspberry Pi 3. The toolchain mentioned at "Prepare ARM Cross Tools" can not be used on a RasPi.
But i found a precompiled toolchain for the Raspberry Pi at https://github.com/vanbwodonk/gcc-arm-embedded-build-armhf. This toolchain is more recent (8-2019-q3-update).
Differing from the description there, i used the following lines to install the toolchain:

wget https://github.com/vanbwodonk/gcc-arm-embedded-build-armhf/raw/master/pkg/gcc-arm-none-eabi-8-2019-q3-update-linux-armv7l.tar.bz2
tar -xvf gcc-arm-none-eabi-8-2019-q3-update-linux-armv7l.tar.bz2
sudo cp -r gcc-arm-none-eabi-8-2019-q3-update/* /usr/local/
sudo rm -r gcc-arm-none-eabi-8-2019-q3-update

Additionally i added the following line at the end of ~/.profile:

PATH=/usr/local/usr/local/arm-none-eabi/bin:$PATH

My first attempt fails to get the toolchain work on my Raspbian Stretch (Debian 9) because the toolchain needs glibc 2.28, but Stretch uses 2.24.
So Raspbian Stretch must be updated to Buster (see https://pimylifeup.com/upgrade-raspbian-stretch-to-raspbian-buster/) to get the toolchain work.

After upgrading Raspbian make runs well and the binary was built.

Hope this helps if someone want to cross compile on a RasPi...

Regards, Thorsten

JulyJim, 2019-11-19:
I have successfully run the current version "remotely " on Raspberry 3B, using Terminal Communication Framework (TCF)  plug-in on Eclipse. 
I have used unmodified Makefile and no other complier / linker modification. 

Since this seems to be ONLY place where developers gather I would like to ask here.If it is OK.

I like to add ability to output debugging messages to standard i/o console.
I do not have NanoVNA hardware and do not intend to actually run the software on it anyway.
 I just want to be able to trace / debug using prinff or cout. 

**I should  be able to add required code to Makefile, but I am not sure how,** 

I'll look into TCF debugging abilities for now.

JulyJim, 2019-11-20:
I am really reluctant to post this here. 
I am finding out that there is NO forum / group dedicated to NanoVNA SOFTWARE development. 
The "development" is scattered on each individual "enterpriser"    using whatever format / forum - if any. 

I am of an opinion that nobody is really interested in NanoVNA as a common project. 
Even this one is focused on fixing issues, which is fine. 
What is really puzzling - the purpose of git(hub) is to share code. 

How can I share code if I cannot physically test it ? 
I have not found a way to run perfectly working NanoVNA code in debug mode. 
I'll keep trying.