hugen79/NanoVNA-H #13: Jog L & R switches appear to stop working after touchscreen menu use

nlroth, 2019-11-06:
I have found a problem with your Oct 9 4-trace release, where I use only the touchscreen and I select CONFIG and then the VERSION menu items and go back up the menu tree to select something else, then back to the version menu and then clear the menu from the screen.
Now, if I try to use the jog switch, I get the menu appear with ENTER but I can no longer get any menu items to highlight. 
It appears the switches are being sensed because if I press Enter to bring up the root menu and then press the Right key 7 times, the menu disappears - but nothing was highlighted.
Thanks.
Larry

hugen79, 2019-11-12:
Let me test it first and it will be improved later.

nlroth, 2019-11-15:
The jog switches appear to have no issue with your latest release. The problem may have been due to an overflow somewhere. You can close

closed 2019-11-15 by nlroth

hugen79, 2019-11-17:
> The jog switches appear to have no issue with your latest release. The problem may have been due to an overflow somewhere. You can close

Trying to replace r5 with a 1k ohm resistor should improve.
hugen