hugen79/NanoVNA-H #15: recipe for target 'build' failed

JulyJim, 2019-11-18:
I am getting this error when running "make" . 

I am not sure what is customary to fix it here.
Is it OK to just report it?  

Description	Resource	Path	Location	Type
recipe for target 'build' failed	rules.mk	/NanoVNA-H/ChibiOS/os/common/startup/ARMCMx/compilers/GCC	line 171	C/C++ Problem
recipe for target 'build/obj' failed	rules.mk	/NanoVNA-H/ChibiOS/os/common/startup/ARMCMx/compilers/GCC	line 178	C/C++ Problem

$(BUILDDIR):
ifneq ($(USE_VERBOSE_COMPILE),yes)
	@echo Compiler Options
	@echo $(CC) -c $(CFLAGS) -I. $(IINCDIR) main.c -o main.o
	@echo
endif
	@mkdir -p $(BUILDDIR)

$(OBJDIR):
	@mkdir -p $(OBJDIR)

hugen79, 2019-11-18:
Please indicate the GCC version and build environment you are using.

JulyJim, 2019-11-19:
Please cancel / disregard my post. 
I had some flaky issues with my IDE and after full reinstall I FINALLY run the software without error , only some irrelevant warnings.

closed 2019-11-19 by JulyJim