hugen79/NanoVNA-H #16: trace command hung

d51r3verse, 2019-11-18:
https://github.com/qrp73/NanoVNA-Q/issues/16
Same issue appeared latest(f6f948b) NanoVNA-H too.

I using empty `trace` command for identify current trace status and scale in my script.
But it will hung occasionally.
I found the only way to remove this issue, `trace x off` then `trace`. Similar issues on `marker` too.

Another issue I found was...
1. Set sweep range to full span 10e3 ~ 1.5e9 / calibrate off
2. `trace all off` -> `marker 1...4 off` -> `pause` (Blank screen / LED steady - For fast fetch)
3. Python loop `scan`&`data 0..1` -> Join&save s2p touchstone via scikit-rf
Exported every `data 0...1` and sp2 files was exactly same.

But data is different with `sweep`&`trace 0..1` enabled.( +4 ~ +20 dB point appeared at CH0 logmag )
NanoVNA requires at least one activated `trace`?

hugen79, 2019-11-18:
Do not use the "pause" command to pause the scan. You can use the "scan" command to pause the screen display and perform a single scan, then use the "data" command to get the data.

d51r3verse, 2019-11-18:
I issued single `pause` command at outside of the loop.
Inside the loop doing `scan` & `data 0...1`, Every iterated data array were different.

In device standalone mode( trace 0 logmag 0 enabled / live sweep ), CH0 logmag can not up to -21 dB.
But, I can see plus level of CH0 logmag when I turned all traces & markers( blank screen / manual scan via PC ).