hugen79/NanoVNA-H #18: SI5351 8mA out causes non-linearity in harmonics mode

erikkaashoek, 2019-11-25:
With 8mA the calibration looks perfect but when measuring a 25 ohm resistor there are some non-linearities. See this post
https://groups.io/g/nanovna-users/message/6595
Suggest to set 6mA maximum. (power 2)

Look for this post to see what performance is possible with your nanoVNA design (I bought from your shop)
https://groups.io/g/nanovna-users/message/6696
This was also done with power 2

hugen79, 2019-11-26:
Thank you erik, I couldn't reproduce what you said on my device and I will test it further. Increasing the output circuit of si5351 will reduce the impedance of ch0, which may reduce the measurement accuracy. The signal input to U6 is far beyond the Ip3 cut-off point of SA612A, which may cause some uncertain situations. I will continue testing.
hugen

erikkaashoek, 2019-11-26:
Hugen,

I redid my measurements. Calibrated with power 1. Then measured a 25 ohm resistor with powers 1,2 and 3 and there are subtle differences.
See these plots
http://athome.kaashoek.com/public/nanoVNA/power%201%2025%20ohm.PNG
http://athome.kaashoek.com/public/nanoVNA/power%202%2025%20ohm.PNG
http://athome.kaashoek.com/public/nanoVNA/power%203%2025%20ohm.PNG

The light blue line is R, it should stay at 25 ohm.
As you can see in the middle of the chart the impedance deviates more and more when the power increases.
its rotated back to eliminate the impact of the electrical length so it should stay at 25ohm. As the highest frequency is rotated to phase zero the electrical distance also changes
Its subtle but it is there and the measurement is best with power 1

erikkaashoek, 2019-11-26:
An even simpler way to see the non-linearities on my nanovna is to measure the uncalibrated SHORT
If the nanoVNA is lineair there should be no difference when measuring on power 1 or power 3 (except for the amount of noise) but there is difference (in my nanoVNA at least)

Pmax65, 2019-11-26:
Hi Erik,
I see the signals are very noisy in the that 500MHz...900MHz range. 
Do you already have done the mods discussed in the link below?
https://github.com/hugen79/NanoVNA-H/issues/14

I use this board to to check the nanoVNA performances:
https://www.amazon.com/gp/product/B081391MJY?pf_rd_p=2d1ab404-3b11-4c97-b3db-48081e145e35&pf_rd_r=W4QNGRADS7JJ65ZT2DBE

After calibrating the nanoVNA with the on-board open, short and 50ohm load, the 33ohm and 75 ohm samples report respectively 33ohm +/-1ohm and 75ohm +/-2ohm.
On the Smith chart and the stimulus set in the range 10kHz-1.5GHz, the 33ohm is almost a perfect dot, while the 75ohm is a little line toward the capacitive area.

Currently, with those mods applied the only issue I see compared to my HP8711A are in the 900MHz-1.5GHz range with high impedance inductive loads where the the Smith chart reports a reduction of the impedance probably due to the input capacitance of the mixers.

Have a great day.

Massimo IK1IZA

erikkaashoek, 2019-11-26:
Did not modify my HW yet, too afraid I might destroy it.....
Most of the noise disappears when I run the nanoVNA from a clean 5 volt supply so I was not concerned about noise.

Pmax65, 2019-11-26:
Hi Erik,
It's not a question of the noise itself, that noise is there because the LO is underdriven at the harmonic range. 
That's could be the reason of the non linearity of the measurement at the harmonic range.
Anyways removing R29, R30, R31, short-circuiting R13 and changing R9 to 200ohm the mixers are correctly driven and the reference input of the Weston bridge is monitored correctly for phase and amplitude.

Have a great day.

Massimo

erikkaashoek, 2019-11-26:
Will try to find some courage to do the modifications.

Pmax65, 2019-11-26:
Hi Erik,
if you never worked with 0402/0603 smt components take care of use a very small tip soldering iron and if you have any friend who already worked on them ask him to help you.
In case the nanoVNA start working crazy (such no longer scan or making very wrong measurements), don't worry to much for that. Most of cases you did a short-circuit on a component, just take a magnifier and search for that.
By the way, the power supply noise can be removed adding few capacitors as explained in that link above, and even the noise coming from the battery charging when the USB is attached to a charger or a PC can be removed using a mosfet, a resistor and a diode.

Have a great day.

Massimo

hugen79, 2019-11-27:
Hello erik, I found a similar phenomenon on the V3.3 version of the hardware. I will consider improving it. The test hardware of the V3.4 version performed well. You can refer to[ issues 14 ]
(https://github.com/hugen79/NanoVNA-H/issues/14)for modification. In addition, Maggie has contacted you and will send you v3.4 hardware next month.

v3.1 hardware test 25 ohm resistor
[image]

v3.3 hardware test 25 ohm resistor
[image]

v3.4 hardware test 25 ohm resistor
[image]

erikkaashoek, 2019-11-27:
Wow, Excellent improvement!!!!.  I will do experiments with the v3.4 above 1.5GHz because I want to measure some components above 2.0GHz
I have already first evidence it is possible to measure the S21 of a 2.0GHz cavity filter but my current nanoVNA HW (v3.1?) is unable to show the S11 as the bridge stops function above 1.2GHz, looks like the v3.4 is solving this problem.

erikkaashoek, 2019-11-27:
For evidence see this post
https://groups.io/g/nanovna-users/message/5149

qrp73, 2019-11-27:
what is the difference in v3.4? Is it possible to do that on v3.1?

hugen79, 2019-11-27:
> what is the difference in v3.4? Is it possible to do that on v3.1?

Hello qrp73, you can refer to [issues 14](https://github.com/hugen79/NanoVNA-H/issues/14) for some modifications, you can send your address to my email hugen@outlook.com, I will send you v3.4 PCB next month.
hugen

Pmax65, 2019-11-27:
hi Erik,

> I have already first evidence it is possible to measure the S21 of a 2.0GHz cavity filter but my current nanoVNA HW (v3.1?) is unable to show the S11 as the bridge stops function above 1.2GHz, looks like the v3.4 is solving this problem.

Wow, aren't you expecting too much from this device? 
;-)
Those 10dB of S/N are not that useful for cavities tuning and I don't believe there is so much margins to get more from those "poor" SA612A mixers, we are already "raping them" :-)

Anyways, I already learnt that with this great device the motto "never say never" is a rule, and I'll look for any other enhancements about this your goal. 

Massimo

Pmax65, 2019-11-27:
Hi Hugen.
once  you have ready the revision 3.4, I would like to buy one to test it. 
Could I buy it directly from you?

Have a great day.

Massimo

hugen79, 2019-11-27:
> Hi Hugen.
> once you have ready the revision 3.4, I would like to buy one to test it.
> Could I buy it directly from you?
> 
> Have a great day.
> 
> Massimo

Hello Pmax65, I already said in [issue 14](https://github.com/hugen79/NanoVNA-H/issues/14) that V3.4 hardware will be sent to you, so you don't need to repeat the purchase.

hugen

erikkaashoek, 2019-11-27:
In the evidence post you will see I got 45dB S/N in S21 at 2GHz with my v3.1 nanoVNA which is sufficient for the tuning.
For best results it is needed to have "some" S11 view. 
See this post for what I tried previously with home build HW for a 2.5GHz cavity
https://erikkaashoek.blogspot.com/2019/02/tuning-25ghz-cavity-filter-with-home.html
seems the v3.4 is breaking the 1.2GHz limit for the SA612 bridge

Pmax65, 2019-11-27:
Hi Hugen,

> Hello Pmax65, I already said in issue 14 that V3.4 hardware will be sent to you, so you don't need to repeat the purchase.

Hoops, I missed that. I'm Italian and my English is not that good, I believed it was about the schematic only.
Thank you very much Hugen.

Massimo

Pmax65, 2019-11-27:
Hi Erik,

> In the evidence post you will see I got 45dB S/N in S21 at 2GHz with my v3.1 nanoVNA which is sufficient for the tuning.

I didn't see it before, but isn't  IAM-81008 a 5GHz double balanced mixer?
I think that SA612A is very less performing above 500MHz than the IAM-81008. 
I was referring to this image from your post:
https://groups.io/g/nanovna-users/attachment/5149/0/2GHz%205V%20power%20amplifier.PNG
Anyways, you must know that when a friend from my HAM radio club proposed me to buy this device this summer, I was a little skeptic on the effectiveness up to the 900MHz, I really marveled to see what I got now up to 1500MHZ.

So, I'll wait for your news.

Have a great day

Massimo

erikkaashoek, 2019-11-27:
The image you refer to is from my v3.1 nanoVNA with some small firmware improvements measuring a 2GHz cavity filter, the measurement was made usable by using a 20dB amplifier AFTER the cavity filter and running the nanoVNA from a clean 5 volt supply. I understand the latter will no longer be needed with the v3.4 HW. And as noise also has been reduced considerably even the amplifier may no longer be needed.

Pmax65, 2019-11-28:
Hi Erik,
I can't warrant you that with rev. 3.4 the post-filter 20dB amplifier will no longer be needed. Since I never checked the modified nanoVNA at 2GHz but only up to 1.5GHz, I don't know the behaviour of the SA612A mixers up there.  Anyways, considering that the dynamic improvement at 1.5GHz for the CH1 port is in the range of 20/30dB depending on the pcb type, I think that it's possible that you can use the new release without that amplifier.

Have a great day.

Massimo