hugen79/NanoVNA-H #20: CH1 port not pure 50 ohm

erikkaashoek, 2019-11-30:
The CH1 input impedance shows a substantial reactive part that hampers measuring certain components, mostly to port devices that are sensitive the the output impedance they see.
This is most obvious in the T-Check where.
Here you see the T-Check performed with only a coax, a Tee and an extra 50ohm resistor
http://athome.kaashoek.com/public/eevblog/NanoTeeCheck.JPG

When you put a 10dB attenuator at the input of CH1 and redo the full calibration and then do the T-Check you get this:
http://athome.kaashoek.com/public/nanoVNA/T-check.PNG
Much improved.
So having a well defined attenuator at the input of CH1 could give substantial improvements because it removes the reactive components of the impedance.

hugen79, 2019-12-02:
Thank you, erik. The input impedance of CH0 is not good. This issue edy555 has been discussed with me. It was also discussed in # 14. The previous circuit did not consider the impact of sa612's input impedance on the circuit. In addition, the line width from the resistance to the sma interface could not meet 50 ohms. I will try to improve it and it will be better in V3.4, but it is still not perfect.

hugen

erikkaashoek, 2019-12-02:
For very demanding measurements I calibrate S21 with a 10dB attenuator before CH1 input. That really helps but you get more noise of course. This is again solvable by adding a low noise 10dB amplifier between the attenuator and the CH1 input
Thanks for taking this into account!

erikkaashoek, 2019-12-02:
Are we talking about the same port? You refer to CH0. I'm talking about CH1

hugen79, 2019-12-02:
Sorry, the previous reply incorrectly, I said ch1.
hugen

hugen79, 2019-12-31:
> The CH1 input impedance shows a substantial reactive part that hampers measuring certain components, mostly two port devices that are sensitive the the output impedance they see.
> This is most obvious in the T-Check
> Here you see the T-Check performed with only a coax, a Tee and an extra 50ohm resistor
> http://athome.kaashoek.com/public/eevblog/NanoTeeCheck.JPG
> 
> When you put a 10dB attenuator at the input of CH1 and redo the full calibration and then do the T-Check you get this:
> http://athome.kaashoek.com/public/nanoVNA/T-check.PNG
> Much improved.
> So having a well defined attenuator at the input of CH1 could give substantial improvements because it removes the reactive components of the impedance.

Hi erik, I have re-simulated the impedance of the CH1 input and the attached test results are new. As the Chinese New Year is approaching, I cannot make a new PCB. I will update the modification and further test in the version after the Chinese New Year.

[image]

erikkaashoek, 2019-12-31:
That seems a substantial improvement! It will help a lot when measuring filters and impedance matching networks as filters can be very sensitive to the impedance connected to their output

PA3CCE, 2020-01-15:
I just ordered the NanoVNA-H4, the above means that the hardware probably is going to be updated again.
Is this change in components only or  in the PCB too, I'm afraid so. 
Components I can change, I updated my 2.8" versions 3.2 and 3.3, to version 3.4 as far as possible. They improved and the display looks more or less like yours above, but more ondulated and the center of the smith chart is a small circle , changing the  PCB I can't ! 
If so I had better wait , now the H4 on his way !,
Jos

hugen79, 2020-01-19:
> I just ordered the NanoVNA-H4, the above means that the hardware probably is going to be updated again.
> Is this change in components only or in the PCB too, I'm afraid so.
> Components I can change, I [...]

Hello Jos
Yes, I will accept the suggestions from the community to update NanoVNA-H. After the update, I will upload the schematic. You may need to change some resistors for better performance.

hugen

PA3CCE, 2020-01-22:
OK, thanks Hugen, in the mean time I received the NanoVNA-H4 , great device, it's loaded with Firmware dated 18//01/2020, here I do find this release only , but I know there is a H release dated  25/01/2020 is this one comparible with H4  or do I need H4 releaases only ?
Jos

PA3CCE, 2020-01-23:
Hugen
Sorry, I was wrong there is no 25/01/2029 release of the firmware, please neglect that.
I understand that in future there will be separate -H an -H4 releases, am I right ?
I hope that I can open the box of the H4 without damaging it, when a resistor change is needed !
Jos

PA3CCE, 2020-01-23:
another error please read 2020, Hi
Jos

hugen79, 2020-03-26:
> Hugen
> Sorry, I was wrong there is no 25/01/2029 release of the firmware, please neglect that.
> I understand that in future there will be separate -H an -H4 releases, am I right ?
> I hope that I can open the box of the H4 without damaging it, when a resistor change is needed !
> Jos

Hello Jos, for nanoVNA-H4 we recommend that you update the firmware after 0.5.0, earlier versions can be problematic.

hugen

hugen79, 2020-03-26:
> That seems a substantial improvement! It will help a lot when measuring filters and impedance matching networks as filters can be very sensitive to the impedance connected to their output

I have uploaded a new schematic, and the impedance of ch1 is closer to pure 50 ohms.
https://github.com/hugen79/NanoVNA-H/blob/F303_ChibiOS_18/doc/Schematic_NanoVNA-H4_2_1.pdf

hugen

electrotoolshobby, 2020-05-18:
hi im new buy nanovna-h online i dont know if , -h product missmatching 50ohm resistance

with u'r last upload schematic have difrent between R34 TX , R49 thru R54 im need to change this to have beter matching sir ? after my vna come, thanks

hugen79, 2020-05-19:
> hi im new buy nanovna-h online i dont know if , -h product missmatching 50ohm resistance
> 
> with u'r last upload schematic have difrent between R34 TX , R49 thru R54 im need to change this to have beter matching sir ? after my vna come, thanks

If you need to accurately measure the devices that are sensitive to the output load, you can refer to the latest circuit diagram modification.

closed 2021-12-11 by hugen79