hugen79/NanoVNA-H #24: Add option to unlock 1.5GHz till 2.1GHz range for those that know what they are doing

erikkaashoek, 2020-01-15:
As the performance of the v3.4 HW is so much better it makes sense to offer an option to unlock the 1.5GHz till 2.1GHz range.
The measurement  results above 1.5GHz have more noise but are usable.
I tested a 1.8GHz low pass filter and a 2GHz cavity filter. Both measured as they should but with some noise.
Could there be a menu option to unlock above 1.5GHz?

x893, 2020-06-12:
Hi,
can you help me with unlock (or link to how to) 1.5GHz ?
Thanks in advance

erikkaashoek, 2020-06-12:
The latest FW by DiSlord is unlocked. Look here: https://github.com/DiSlord/NanoVNA

x893, 2020-06-12:
Thanks, i will try.

x893, 2020-06-12:
DiSlord repo looks interesting but i can't found STM32F3 text,
I have NanoVNA-H4 (FW 0.5.0) with HW version 4.2.