hugen79/NanoVNA-H #25: NanoVNA-H4 center frequency change above 1 GHz

hwalker6742, 2020-01-22:
hugen please see my review of the NanoVNA-H4 at https://groups.io/g/nanovna-users/message/10012 .  I encountered the lock-up when trying to set the center frequency in the 1 - 1.5 GHZ range.  Another member, https://groups.io/g/nanovna-users/message/10024 , also confirmed his NanoVNA-H4 locked up when setting the center frequency in the same range.

Multiple users have also encountered a QA problem, https://groups.io/g/nanovna-users/message/9914 , where the USB-C case opening is too small and the cable will not fit.  My unit was fine but a few members have had to enlarge the opening by hand.

- Herb

hwalker6742, 2020-01-23:
Additional feedback on the center frequency issue was provide in https://groups.io/g/nanovna-users/message/10088 .  Lock-up requiring power cycling occurred when setting center frequency above 1 GHz.  Lockup does not occur when setting start and stop frequencies.  Updating to the Jan 18, 2020 firmware did not resolve the issue.

- Herb

hugen79, 2020-01-24:
Thank you for your feedback, this is a serious problem, I haven't found the cause yet, I will look for the cause further.

hugen

Pmax65, 2020-01-24:
Hi Herb,

> Additional feedback on the center frequency issue was provide in https://groups.io/g/nanovna-users/message/10088 . Lock-up requiring power cycling occurred when setting center frequency above 1 GHz. [...]

I checked the HW3.4 and the H4 4.2 devices with the original FW and they work right. 

I also tried "bad" settings like setting the center frequency to 1.5GHz and the device correctly switched to CW and after that I set the span to 200MHz and I got a correct 1.4GHz CF.
Anyways, I never had any lock-up.

Have you tried to clear the whole memory before downloading the update?

Massimo

BTW, Hugen I finally got the damned Eclipse IDE working on one PC but it is still not compiling on my home workshop main PC. 
This is my first approach to the STM32 family indeed and I honestly never experienced a so complicated to setup tool in my professional experience (ok, it could be because I almost always used not free IDE with their own integrated compiler/debugger). 
One curious behaviour is that even on the PC which performs the compiling task right, the real-time parser still report some undefined constants which remain undefined (just for the parser) despite they are well defined in the assembler list produced by the compiler.

hwalker6742, 2020-01-24:
Pmax,
I have not performed any updates (see attachment).  Two other users have provided feedback at groups.io of their NanoVNA-H4’s locking up when setting the center frequency above 1.5 GHz, so it is not just my unit.  So far you are the only person who has said they do not have this problem.  One user did update their NanoVNA-H4 but said that the center frequency lock up problem still remained. 

One strange thing about my unit is that if I set the range for 10k – 1 GHz and select  center frequency. The default setting is 500 MHz as expected  (see attachment).  But if I set the range to 1 – 1.5 GHz and select center frequency.  The default setting is a negative number.

Setting the start and stop frequencies is fine.  It is only setting the center frequency that locks up the unit and requires cycling power.  The lock up also stops USB communications so I could not reset the unit via the CLI.

Windows10 x64, i7, 12GB
- Herb

[image]
[image]
[image]

Pmax65, 2020-01-25:
Hi Herb,
I apologize.
Since I was checking both the models, I believed to having performed that checks also on the H4 model. After you wrote the message above I decided to re-check the H4 (probably "check" instead of "recheck" indeed) and effectively there is the problem also after the last update.

Pmax65, 2020-02-02:
Hi Hugen,
despite I'm not able to avoid those errors from the Eclipse parser, I've been able to debug this issue.

There is a too small string buffer into the [ void draw_frequencies(void) ] function in Plot.c file
the string buffer buf() is only 24 characters long, while the code sequence: 
        strcpy(buf, "CENTER ");
        frequency_string(buf+7, 24-7, fcenter);
        strcat(buf, "    ");
Writes these characters: 7 for the word CENTER + one space, 16 for the frequency in MHz when it's above or equal 1GHz and 4  for the ending spaces.

Summing them the buffer is filled with 27 characters + the \0 it occupies 28 bytes non 24 as reserved into the buffer definition.
Note that also all the other frequencies printing probably corrupt some memory location.

And more, it probably corrupt different memory from the 2.8" version, but this issue is present also there and it could be the reason that very sporadically I had some hangups of the instrument also in the 2.8" type.
Note that fixing this issue I never had the issues I reported here https://github.com/hugen79/NanoVNA-H/issues/27#issue-555908752
Maybe that this issue was the culprit of many sporadic bad behaviours of the two instruments.

I finally found one more issue:
there is a bug in computing the frequency when setting the center frequency with the span already set to 1.499GHz. If the center frequency is set with the sum of the center frequency + span/2 greater of 2.147GHz the computation fails and leaves the span as it was before. Anyways this is because a 32 bit signed integer can hold a maximum value of 2^31=2,147,483,648 and during the frequency0 and frequency1 computation, this limit it's overcome. You should adjust that function to avoid that.

Have a great day.

I look for your next update.

Massimo IK1IZA

closed 2020-02-05 by hwalker6742