hugen79/NanoVNA-H #32: Add Impedance /Z/   to display/format  menu

UlrichD61, 2020-02-15:
Hello,
I'm just asking whether it would be possible to add to the NanoVNA menu the "absolute impedance" /Z/ as well, rather than only resistance (real and imag) separately. 
Real and Imag part is important, for sure, but for some measurements it would simplify to also display /Z/ as well and not having to use Pythagoras every time. 

I'm using standard 2.8" NaonVNA primarily for mesuring impedances of ferrites in S0 reflection mode. works perfect !

Thanks a lot for all the SW/FW/HW work in the past !!!
Ulrich

hugen79, 2020-02-17:
OK, if there is enough memory, I will add it.

UlrichD61, 2020-02-17:
Hi, thanks a lot. 
If helpful I can "try and error" your SW version before you release it. 
That /Z/  feature would be very helpful for me when presenting some impedance results to my "audience" 

Cheers
Ulrich