hugen79/NanoVNA-H #37: My old fixes for NanoVNA-H

DiSlord, 2020-03-12:
Hi after i by NanoVNA-H i begin fix from this firmware.

But not public it, and adaptate to eddy (and continue work on it), but old font fixes still exist

I upload this fixes to my repo https://github.com/DiSlord/NanoVNA-H
Not test, but must work

Possible in future i order nanoVNA-H4 and try work on it

hugen79, 2020-03-26:
Hello DiSlord , i'm sorry to reply to your message late. I am following your modifications in the original branch of TTRFTECH. I tested your estimate at https://github.com/DiSlord/NanoVNA-H, the typography is better, I will try to use your new code and merge the code you updated on ttrftech. 

If you need nanoVNA-H4, you don't need to buy, please send your address to my email hugen@outlook.com, I will send it to you. Please include your mobile phone number when sending the address, sending international express must fill in the mobile phone number, please understand, thank you!

hugen
2020-03-26

DiSlord, 2020-03-27:
Thanks, i write you on email

Possibly better port all code from original branch of TTRFTECH, i try write code more easy change font size and resolution. I think no big problem port it and use bigger font from github.com/DiSlord/NanoVNA-H, but i don` know how fast work screen update on bigger screen size (render smith grid  so slow). Other code should work.