hugen79/NanoVNA-H #38: Unexplained behaviour of calibration correction around high concentration of poles and zeros

Pmax65, 2020-03-12:
Hi Hugen.
I'm working about the odd harmonics loss of dynamic when a LPF filter let pass signal in the fundamental band below 300MHz and cuts in the odd harmonics bands above 300MHz.
I already designed a 5th order elliptic HP filter and modified one nanoVNA to clock the MPU and the DSP with a separate 8MHz oscillator as per your nanoVNA-H4 model, and used the CLK2 output of the Si5351A to drive the HP filter to get a cleaner signal with the fundamentals below the harmonics level when the DUT is checked above the 300MHz limit. 
I modified the firmware to get the test signal that drives the DUT directly from CLK1 for frequencies below 300MHz and to get the test signal that drives the DUT from CLK2 passed through the filter for frequencies above. 
Now the dynamic improved a lot making the device useful for duplexer testing/tuning.
Doing that I experienced an unexplained (for me) behavior of the calibration function of the S11 plot in the range between 150...350MHz. 
Since for the moment I simply paralleled the filtered CLK2 output to CLK1 output, this last is influenced by the HP filter poles and zeros in that range (it's an elliptic filter and in the stop band it has abrupt changes in impedance there). I already planned to put an RF switch to remove this loading issue, but I would point out this issue anyways because it seems that the interpolated correction works bad.

**This is the uncorrected plot of that range:**
[image]
**While this is the plot of the same range when corrected by the interpolated calibration parameters:**
[image]
I did the 10kHz... 1.5GHz full span calibration and the marker in the plot are placed in the point of that full band calibration. 
My question is: 
how could the calibrated plot showing those almost parabolic interpolation between two consecutive calibration points?
I also can't explain why that red marker is so high, since this is a calibration point computed exactly with the very same dummy load used for the plot. Anyways, how could the S11 between two successive calibration points at lower RL have an higher RL?
This happens only when the DUT is well matched, in case of S11 close to 0dB this issue disappear little by little as the 0dB line is approached.

Do you have any idea about this?

Have a great day.

Massimo - IK1IZA

qrp73, 2020-03-13:
There is some "simplification" in si5351 frequency setup code, it leads to wrong frequency set at some frequency boundaries. Like 300 MHz, 600 MHz, etc. And your issue looks like that, but the errors are propagated due to multi-sweep in the NanoVNA-Saver.

It's hard to fix, because such fix requires complicated logic. But you can avoid it. Just don't use sweep through these frequency boundaries, if you can't, then setup a small frequency step, it will minimize amount of points with a wrong frequency.

DiSlord, 2020-03-13:
Its problem come from set si5351 power(in auto mode, power = -1), try set power 0 or 2 and calibrate on all diapasone, and check again.
Also need correct gain table

Pmax65, 2020-03-13:
Hi to both (qrp73 and DiSlord),
First of all, thank you for you race about this.
Maybe I've been not clear in last message, let me try to tell you what is happening in other words.
The two plot plots are the results of tests of the same DUT using the same nanoVNA, but while the first is with the correction disabled (not resetted) the second is with the correction enabled. 
In few words, I can switch between them just using [CAL]>[CORRECTION].
Please keep in mind that the calibration was done from 10kHz to 1.5GHz so the real calibration points are placed with 15MHz spaces between them all the other points are the results of the interpolation of the 2 consecutive calibration points.
The 5th order elliptic high pass filter is connected as per the following schematic detail:
[image]
As you can see, while when the frequency is above 300MHz the signal exits from CLK2 passing through the filter and sees the load of CH1 which is almost a resistive load with a little parasitic capacitance, when the frequency is below 300MHz the signal exits from CLK1 and sees the filter output via FC3. The discontinuities seen in the top plot is due to the poles and zeros in that range of frequency (excluding the 300MHz discontinuity of course, which is due to the band change).
I can easily remove that discontinuity by an RF SPDT switch, but I can't imagine why the calibration algorithm introduces that artifacts  between the calibration points. Those almost parabolic correction shouldn't exist. Incredibly it seems that the correction works the other way it should work. In fact if I traced straight lines between those calibration points I have the real frequency response of the DUT S11 (except for the red calibration point which is completely wrong placed).
It seems to me that there could be a very bad bug in the calibration algorithm.

DiSlord, 2020-03-13:
Data on intepolation mode - use linear function see cal_interpolate(int s)
as write eddy https://github.com/ttrftech/NanoVNA/issues/73

DiSlord, 2020-03-16:
Problem come from here:
```
        // avoid glitch between freqs in different harmonics mode
        if (IS_HARMONIC_MODE(src->_frequencies[j]) != IS_HARMONIC_MODE(src->_frequencies[j+1])) {
          // assume f[j] < f[j+1]
          k1 = IS_HARMONIC_MODE(f) ? 1.0 : 0.0;
        }

        float k0 = 1.0 - k1;
        for (eterm = 0; eterm < 5; eterm++) {
          cal_data[eterm][i][0] = src->_cal_data[eterm][j][0] * k0 + src->_cal_data[eterm][j+1][0] * k1;
          cal_data[eterm][i][1] = src->_cal_data[eterm][j][1] * k0 + src->_cal_data[eterm][j+1][1] * k1;
        }
```
Linear interpolation in most cases work good, but then change calibration from one harmonic to other:
```
        // avoid glitch between freqs in different harmonics mode
        if (IS_HARMONIC_MODE(src->_frequencies[j]) != IS_HARMONIC_MODE(src->_frequencies[j+1])) {
          // assume f[j] < f[j+1]
          k1 = IS_HARMONIC_MODE(f) ? 1.0 : 0.0;
        }
```
Its non correct. Used last value from left or first from right (not do interpolation), need made extrapolate use 2 point before or 2 after
Also it not work on 900MHz (used IS_HARMONIC_MODE(f)), it only for 300, for 900 used point after

Pmax65, 2020-03-16:
Hi DisLord.
I already fixed this while I added the RF switch to the schematic. 
I implemented a simple extrapolation algorithm computed on the two closest measured  calibration points, instead of the interpolation at the two edge of the band change point.   
It works great at 300MHz, but the IS_HARMONIC _MODE macro is fails to detect the other band changes (the 900MHz) probably because it has been forgiven when the instrument has been extended above 900MHz.
This is my

closed 2020-03-16 by Pmax65