hugen79/NanoVNA-H #42: Reference impedance setting

goscickiw, 2020-06-01:
Sometimes I need to measure a 75 ohm system, but 75 ohms shows as 50 ohms on the graphs when I calibrate using a 75 ohm load. Would it be possible to add a setting for impedance in the software, so I can match it to the resistance I use when calibrating?

I understand that anything other than 50 ohms is most likely not optimal for the NanoVNA's hardware, and that it would be batter to use something like impedance converters, but right now I don't need anything too precise and I haven't encountered any big discrepancies when measuring in 75 ohm, other than the displayed values being 2/3 of the actual value.

There could also be a warning when changing the impedance value, that it could make the results less accurate.

hugen79, 2020-06-03:
In order to obtain good results, you need to modify the reflection bridge, then modify the software, and finally use a 75 ohm load for calibration.

goscickiw, 2020-06-03:
Yes, I know this, I mentioned that previously. If calibrating the whole system to 75 ohms is causing an issue that I haven't noticed yet, maybe the software impedance value could only affect the SWR (1:1 at 75 ohms) and Smith chart (center at 75 ohms) while the calibration is still to 50 ohms?

Also, how exactly would the reflection bridge have to be changed to accept both 50 and 75 ohm calibration?

Pmax65, 2020-06-03:
Hi goscickiw,
> Yes, I know this, I mentioned that previously. If calibrating the whole system to 75 ohms is causing an issue that I haven't noticed yet, maybe the software impedance value could only affect the SWR (1:1 at 75 ohms) and Smith chart (center at 75 ohms) while the calibration is still to 50 ohms?

No, is not only that, in case you don't apply a 50/75ohm adapter on CH0 during the calibration and the measurements, you get a flatness in the measurements that for the impedance offset could be up to 3.5dB in the return loss measures. 
Note that there is no need to rebuild the whole Weston bridge to get a right measurements, it suffice a 50/75ohms minimum loss pad between the CH0 port and the DUT. 

The only side effect is that the adapter reduces a little the dynamic of the measurements, but for RL shouldn't be a tragedy at all. 

Have a nice day.

Massimo - IK1IZA

goscickiw, 2020-06-03:
Even with 50 ohm calibration and a converter, the 75 ohms (in system) will be displayed as 50 ohms (what the VNA sees at the converter), and 1:1 SWR and the center of the Smith chart will still be 50+0j ohms. I would like to be able to set 75 ohms as Z_0, only visually for proper values on the graphs.

So the 5.7 dB loss on the converter's resistors won't be a big issue? Currently I have some TP-75 transformers which have low loss but they only work properly up to 200 MHz, and I heard their inductive nature can affect the measurement.

Pmax65, 2020-06-03:
Hi goscickiw,
yes, if you place a minimum loss pad you get right measurements at cost of the reduced dynamic of the port. This is because it is a pure resistive (non-reactive) device in the useful range of frequency  for which it is designed, so the calibration can easily embed its behaviour.
In that case you need to change the SW for center the Smith chart at Z0=75ohm of course.
I never tried the Macom TP-75 transformers for this scope, but my guess is that its reactance along the frequency range could be not fully predicted and for that I suppose that the calibration shouldn't be effective in this case. 
Please, take care that this last is just an opinion of mine, I never checked it out.

Massimo - IK1IZA

Pmax65, 2020-06-03:
By the way (this is just a provocation of course). 
Hugen could design and sold cheap SMA male to BNC female minimum loss pads for the nanoVNA. 

They are just one 43.3ohm resistor in series to BNC hot terminal and an 86.6ohm resistor in parallel to the SMA male connector   ;-)