hugen79/NanoVNA-H #46: NanoVNA bricked after installing latest update

SheetLightning, 2020-07-27:
First time every updating the firmware on this. I checked I had the -H version and the firmware version message pointed me to this GitHub. Unfortunately my NanoVNA booted to a white screen after installing the latest update: NanoVNA-H4_20200221.bin

Mine is the -H version and I am updating from a PC running Linux Mint 19.2 so I used the instructions from here:

https://nt7s.com/2019/11/updating-the-nanovna-firmware/

Also here:

https://oristopo.github.io/nVhelp/

Used the following command line:

dfu-util -d 0483:df11 -a 0 -s 0x08000000:leave -D NanoVNA-H4_20200221.bin

My screen came up white and nothing else happened. I have tried bridging the link to get into DFU mode but the screen remained white.

After a bit of a panic moment I tried uploading the firmware again anyway and to my surprise it worked in the background - it was just that the display was not telling me the NanoVNA was in DFU mode. Uploading again got the a white screen again, so I tried the ch.dfu from edy55. This uploaded and worked OK, except my SWR readings were not right (yes I did calibrate).

After a bit of further digging, I noticed that in the previous version of your firmware (18th January) there is an version with an AA suffix - file NanoVNA-H_AA_20200118.bin. Since my unit originally had four traces I tried uploading this to the unit and it did work!  At least the unit is now working with the -H_AA firmware version 0.4.5-4-g96e7efe. I have yet to test the SWR readings, but could I request that the "AA" suffix be added to the version number in the version message to distinguish this firmware?

I am not sure why the latest _H version does not boot, but the previous -H_AA version works fine.

Hopefully when the AA version of the latest firmware is released I will be able to apply that to this unit, but in the meantime, thank you for your work on this firmware.

Pmax65, 2020-07-27:
Hi SheetLightning,
if your nanoVNA-H has a 2.8" display you must use the nanoVNA-H firmware not the nanoVNA-H4 one. The latest is for the 4" inch display version that uses the STM32F330 MCU instead of the STM32F072 installed in your device. for this reason you device is now locked.

Just download the right MCU firmware and you'll get back the device working.

By the way, since the device is hanged-up now, you must turn on the board while shortcutting the two P1 pins named Boot and VDD to start the DFU mode.

SheetLightning, 2020-07-29:
Thank you for your reply. That explains it. Looking at the releases the 18th Jan 2020 release is a -H release which is why going to the earlier release worked. The later 21st Feb 2020 release changed to -H4 but I had no idea of the reason for this change so thank you for the explanation. I managed to turn on DFU mode using the jumper on P1 as described. Thank you.

evgsm1979, 2020-10-18:
Доброго времени всем кто читает. Приобрёл NanoVNA-H v.3.4 и при первом включение получил белый экран. При втором включении прибор заработал, я подумал, что это что-то с прошивкой и перепрошил на более свежую. Но даже после этого периодически включался с белым экраном. Прошивал прошивками для H версии. Теперь прибор включается только с белым экраном. Светодиод работы мигает, но изображения нет, в режим DFU переключал закорачиванием контактов на плате. NanoVNA Saver c прибором работает, значит процессор целый. Может неисправна TLV320? Как заставить экран заработать?

Pmax65, 2020-10-19:
Hi evgsm1979,
it looks like there are some odds on the LCD connections.
Unluckily it is soldered on the other side and the LCD metallic back shield is glued to the pcb.
The only way to redo the soldering is to unlatch the shield clips as seen in this link:

https://1.bp.blogspot.com/-oQ5HUbjfH6o/XUhfoaIoAFI/AAAAAAAAEYM/R4ZhjcC1VxEfbgxrmp-qSZgWoxZG6AZsQCLcBGAs/s1600/NanoVna%2Btop%2Bguts.jpg

Check for the STM32F072 solderings too, especially for those by which the LCD is connected with.
I don't believe the TLV320 has something to do with this issue.

Have a nice day.

Massimo

evgsm1979, 2020-10-19:
Good afternoon, Massimo. Thank you for your very quick response. I soldered the screen again. Not help. I will solder STM32F072 again. But I noticed that the device freezes in front of the white screen. The device often freezes and then turns on a white screen. The device itself works with a PC and Android, only the screen is white. This is quite inconvenient, I need to set up antennas with it. Stitched also always good versions for n. Good day UB9YWB 73!!!

evgsm1979, 2020-10-19:
Or maybe it's just a defective screen, how can I check it to find out for sure?

evgsm1979, 2020-10-19:
https://drive.google.com/file/d/15P-SOFXcuvib-plJeht2BXfw019qVZhz/view?usp=sharing,%20https://drive.google.com/file/d/165TWeE2D9kyg9Uqo5cQSdkoxXqVeGtmi/view?usp=sharing,%20https://drive.google.com/file/d/16FaPWjqLuECJDb-6csWQGunYUNd_UoYC/view?usp=sharing,%20https://drive.google.com/file/d/16ObS7pwfoOdGVYnoRpnB5T74zdlBkzJY/view?usp=sharing

Pmax65, 2020-10-19:
Hi evgsm1979,
is the above the picture of your own LCD?
I don't know if the lower side of the LCD flexible pcb has a twin contact for any upper contact, but ground pin 8 looks broken in that picture.
Some LCDs have separated ground for the I/O controller, the LCD and the serial interface. 
In case that it's your LCD, I'll tried to redo that pin8 connection.

Have a great day

Best 73 from Massimo - IK1IZA

evgsm1979, 2020-10-19:
Hello again. Photos of the display are mine. I took it off and re-soldered it. They call all the contacts everywhere. Also soldered STM and TLV320. There are no changes. The screen doesn't work, only the backlight is on. I'll attach a photo. What should I do next? Thank you for your answers.
https://drive.google.com/file/d/1k5z7e_XZYNlquH9XwBhgbrxOsQyBSUFM/view?usp=sharing

Pmax65, 2020-10-20:
Hi evgsm1979,
everything lead to a defective LCD, but since you said that "The second time the device was turned on", I guess that you meant that the LCD at least one time worked right, is it that?
In that case, if you were lucky, you could redo all the soldering of the components on the flexible pcb. 
Between them, there are the serial interface timings capacitors, maybe that one bad soldering inhibits the serial interface to run correctly.

Just try it, and let me know.

In the meantime, have a great day.

Massimo - IK1IZA

evgsm1979, 2020-10-20:
Massimo good afternoon. Capacitors on a flexible loop soldered. It didn't help. Stood up with an oscilloscope on the display contacts and signals and go to it. So the display is probably faulty. Thank you for your assistance in Troubleshooting. You need to buy a suitable display on aliexpress, I live in the village and we do not have radio stores. Good Luck To You 73. UB9YWB.

Pmax65, 2020-10-20:
Hi evgsm1979,
sad to read that. 

Those LCDs are COGs and sometimes the slight difference in expansion between the silicon chip and the glass may lead to connection disjuncture. 

Here is a research about that issue.
https://www.jim.or.jp/journal/e/pdf3/52/11/2106.pdf

It looks like you had been unlucky.

Have grat days, 
Massimo

evgsm1979, 2020-10-21:
Hi, Massimo. Well, then just bad luck with this instance. I will order a new one in China. Thank you for your help in solving the problem. Good luck in life and business, the topic is not closed yet, I will write the result on the screen. UB9YWB 73!

333,86 руб. | 3,5-дюймовый TFT ЖК-экран
https://a.aliexpress.com/_AX6C9l

Pmax65, 2020-10-21:
Hi evgsm1979,

luckily it looks pretty cheap, so you have just to wait for its arrival.
Let me know how it will work.

In the meantime have great days.

All the best from Italy.

Massimo - IK1IZA

evgsm1979, 2020-10-21:
Greetings from Siberia. You can wait until November 20. As soon as I install it, I will definitely write here. 73! UB9YWB.

evgsm1979, 2020-11-21:
Hi, Massimo. Got a screen with Aliexpress. Set and the instrument is earned. So it was the display. All the best, health, take care of yourself. 73!!! UB9YWB.

Pmax65, 2020-11-21:
Hi evgsm1979,
nice to read that you fully fixed the issue with little bucks and just some time to wait.
I hope you enjoy your nanoVNA.
Have great days.
I wish you all the best-

73 de IK1IZA

schumar, 2021-09-20:
Just in case someone found this report because they were searching for the "does not match device" error:
It helps to simply *not* specify `-d 0483:df11` when flashing, i.e. just do
```sh
dfu-util -a 0 -D NanoVNA-H_20210916.dfu
```