hugen79/NanoVNA-H #47: Is it normal for the readouts to be this low when uncalibrated?

goscickiw, 2020-09-05:
I have been using the NanoVNA-H for some time now, and I've been wondering this for a while, but never cared about it too much as it worked fine when calibrated. However, recently I bought a S-A-A-2 and it doesn't seem to have this problem when uncalibrated, so I finally decided to ask.

The uncalibrated magnitude measurements for CH0 are about -20 dB when nothing is connected, and for CH1 they're -10 dB when CH0 and CH1 are connected together with coax. This seems oddly low, as in both cases they should be close to 0 dB. I remember that it didn't have this issue out of the box, but it might have been pre-calibrated by the seller, and I did a firmware upgrade shortly after receiving it which most likely removed the original calibration.

This is how it looks like when not calibrated:
[image]

With calibration for comparison:
[image]

Is it normal, or does it mean that it's defective, or a crappy clone? (I didn't buy it from the official alibaba link, but from a local seller [here](https://www.konektor5000.pl/index.php?p4509,nanovna-h-najmniejszy-analizator-antenowy-50khz-1500mhz-dotykowy-wyswietlacz-2-8-najnowsza-wersja-h))

hugen79, 2020-09-10:
Yes, the original NanoVNA design has this problem. The reference signal is too high, resulting in low uncalibrated data. This problem can be solved by calibration. SAA2 uses a different design.

goscickiw, 2020-09-10:
Thanks, looks like my NanoVNA is fine then.

closed 2020-09-10 by goscickiw