hugen79/NanoVNA-H #51: White screen of NanoVNA-H 3.4

evgsm1979, 2020-10-19:
Good time to all who read. I purchased a NanoVNA-H V. 3.4 and got a white screen when I turned it on for the first time. The second time the device was turned on, I thought it was something with the firmware and flashed it to a more recent one. But even after that, it periodically turned on with a white screen. Flashed firmware for the H version. Now the device turns on only with a white screen. The operation led is blinking, but there is no image. I switched to DFU mode by shorting the contacts on the Board. NanoVNA Saver with the device works, so the processor is whole. Is the TLV320 faulty? How do I make the screen work?

Pmax65, 2020-10-19:
I just replied you here:
https://github.com/hugen79/NanoVNA-H/issues/46#issuecomment-711723428

slack-t, 2024-03-26:
If anyone runs into this error again. I yesterday encountered the same issue. Suddenly I got a white screen after switching the device on.
Before trying to replace the screen, I just reflashed the latest firmware using STM32CubeProgrammer. I bridged Boot0 and VDD, switched the device on, confirmed it's in bootloader mode using windows device manager and used the following settings:

[image]

restarted and now my nanovna works again!