hugen79/NanoVNA-H #53: PPM offset in config menu

Joel-Mckay, 2021-02-21:
Updated to: NanoVNA-H_20210130.dfu
10MHz hc49 xtal test reads as 9.997225MHz

Can we use a GPSDO 10MHz reference to add the ppm offset config into the nanoVNA-H menus?
Would seem kind of odd to have to have to flash a custom compiled firmware after warm up, as some are suggesting.

A "cal ppm err @10MHz" option would be stellar ;-)

Pmax65, 2021-02-22:
Hi Joel,
I don't want to teach something that you probably know better than me, but that offset is far too high to justify a bad clock compliance. The nanoVNA uses a good low ppm reference, while the value you reported is on the order of 2775ppm (!!!), which is worse than any even cheapest xtal on the market.
Are you doing the pragmatic measurements for a xtal by means of a VNA?

In case you don't, take a look at this nice VE6WGM's video :

https://www.youtube.com/watch?v=rVIHVi-7brs

Have a great day

Massimo - IK1IZA

Joel-Mckay, 2021-02-22:
Hi Massimo,
Thank you for the reply =)

Yes, I have done quite a bit reading on using the nanoVNA to profile xtals.  In general, I found this method comparison a better video, as it illustrates ways to verify results.
https://www.youtube.com/watch?v=G9zZRNzhsEE

Like yourself, I was rather surprised at the ppm error I ran into. Initially, I thought the manual method I used was in error, grabbed the wrong hc49, or maybe my test jig was wrong. I also tried debugging a little script the community is working on to help improve the laborious matching process, and a similar error occurs with several vendor xtal samples (all report low and within 200ppm).
 
What signal should we expect if  the stimulus output is set to sweep 10.0MHz to 10.0MHz?
 (i.e. if your unit reports higher than mine, than I may have to probe my PCB oscillator is near 26MHz. if that checks out, than I can roll back to an earlier stable firmware to exclude a possible bug. A version  recommendation would be appreciated.)

Cheers,
Joel
VE7NTP

Pmax65, 2021-02-22:
Hi Joel,

I'm not sure, what do you mean with: "What signal should we expect if the stimulus output is set to sweep 10.0MHz to 10.0MHz?"
Is that a zero span scan?

Anyways, the resonating frequency of a xtal depends upon the capacitive loads applied to its terminals, referring to the following link second drawing, the xtal load capacitance "pull" the resonating frequency between Fs and Fa (a little less indeed, of course)
https://www.microsemi.com/blog/2016/12/15/crystal-pullability-series-vs-parallel-resonant-crystals/
For that reason it is possible, if not probable that the nominal frequency of the xtal you are measuring is obtained with a specific load capacitance applied at its terminals and the ground. 
Usually xtals for logic oscillators need those capacitances in the range of 10 to 40pF to get the nominal frequency marked on them.
Could it be that your issue?

Have a great day.

Massimo - IK1IZA

P.S. 
I hope I've been clear above, I apologize for my English.

Pmax65, 2021-02-22:
By the way,
take a look at this this Keysight application note:
http://literature.cdn.keysight.com/litweb/pdf/5965-4972E.pdf

In last paragraph figure 13, you can see a specific xtal test fixture to get accurate measurements by null all the parasitic behaviors  of the cables, since very low capacitance can offset the xtal resonance.

Joel-Mckay, 2021-02-22:
"I'm not sure, what do you mean with: "What signal should we expect if the stimulus output is set to sweep 10.0MHz to 10.0MHz?"
Is that a zero span scan?"
Yes, I can measure the single frequency output against a very precise reference. However,  VCO/PLL frequency synthesis are never perfect, and knowing what it should be helps.  i.e. if it is my 50ohm strip-line measurement setup, than the  approx 10MHz read with your scope or counter will still match my nanoVNA output.  Removes a lot of variables.

Thanks again for the links, I will read them when I get home. 

Your English is still much better than my Italian...  ;-)
73
VE7NTP

Pmax65, 2021-02-22:
Hi Joel,
now I'm at my home workshop and I made what you said, that is I connected a SMA T to the S11 port and to my Rohded & Schwarz SME generator driven by my DGPSO system on the symmetric port of the T then I connected the S21 port the third T connection via a short cable. I set the nano VNA to 10MHz CW and the generator to 10.000100MHz and I adjusted the generator  level to get the maximum beat on the nanoVNA S21 trace. Then I tuned the generator for the lowest beat frequency (that is for the slowest up and down trace movement).
My nanoVNA reported the minimum frequency beat at -0.2ppm (9.999998MHz)
My nanoVNA H4  reported the minimum frequency beat at +0.7ppm (10.000007MHz)
Hope this could help you.

Have a nice day.

Massimo - IK1IZA

DiSlord, 2022-02-01:
You can calibrate nanoVNA XTAL, if nano know real internal XTAL frequency - it allow correct output.
Now go to
CONFIG->EXPERT SETTINGS->TCXO and set default 26MHz
Set CW freq to 26MHz, wait while device warm up (near 5-10min) measure output frequency.
CONFIG->EXPERT SETTINGS-> TCXO and iput this freq.
After need save config (CONFIG->SAVE)

After this correction NanoVNA CW output must be equal as enter (exist only ~1-2Hz drift, and while device warmup near 10-20Hz)