hugen79/NanoVNA-H #57: Reload FW after uploading wrong version

DABtester, 2021-05-24:
Accidently i uploaded the H4 firmware, and now my device is bricked.
Any way to un-brick it again / upload the correct version?

Thanks

Pmax65, 2021-05-24:
Hi DABtester.
Since the device is hanged-up now, you must turn on the board while short-cutting the two P1 pins named Boot and VDD to start the DFU mode.

Have a great day.

Massimo - IK1IZA

benner, 2021-05-24:
IIRC it can be done by holding some buttons when powering up  too

Pmax65, 2021-05-25:
Hi benner,
> IIRC it can be done by holding some buttons when powering up too
That depends upon the previously uploaded firmware. He got the hardware hanged up by a wrong firmware upload, it shouldn't work because the device is locked somewhere unknown in the code.
The P1 short-cutting is embedded in the bootstrap code and it should be always available independently by downloaded firmware.

Have a great day.

Massimo - IK1IZA

DABtester, 2021-05-25:
> Hi DABtester.
> Since the device is hanged-up now, you must turn on the board while short-cutting the two P1 pins named Boot and VDD to start the DFU mode.
> 
> Have a great day.
> 
> Massimo - IK1IZA

That worked - many thanks for your help!

closed 2021-05-25 by DABtester

Pmax65, 2021-05-25:
Nice to be helpful to you.

Have a great day.

Massimo