hugen79/NanoVNA-H #59: firmware make error

JP-rg, 2021-08-30:
I have tried a number of times and looked at editing the files as stated but no success I get the following error

Compiling hal_lld.c
ChibiOS/os/hal/ports/STM32/STM32F0xx/hal_lld.c:54:40: error: expected identifier or '(' before 'void'
 static void hal_lld_backup_domain_init(void) {
                                        ^~~~
ChibiOS/os/hal/ports/STM32/STM32F0xx/hal_lld.c: In function 'hal_lld_init':
ChibiOS/os/hal/ports/STM32/STM32F0xx/hal_lld.c:235:30: error: expected expression before ')' token
   hal_lld_backup_domain_init();
                              ^
make: *** [ChibiOS/os/common/startup/ARMCMx/compilers/GCC/rules.mk:216: build/obj/hal_lld.o] Error 1

Please advise

jovo3, 2021-11-04:
Same here. I stepped back in the versions and am able to build until 0.5.0. But once at version 1.x.x, I get the same error.
I have also built the examples in ChibiOS for STM32 and they are building just fine. Any clues here?

PeterDams, 2021-11-04:
I have the same problem, searched everywhere but no solution. This issue is already open since aug 31th. Hope to see an answer soon ;o)

jovo3, 2021-11-07:
Here is a docker file to reproduce the issue we are seeing:
FROM ubuntu:latest

RUN apt -y update
RUN apt -y install git wget make dfu-util

RUN git clone https://github.com/hugen79/NanoVNA-H.git
RUN cd NanoVNA-H/ && git submodule update --init --recursive

RUN wget https://developer.arm.com/-/media/Files/downloads/gnu-rm/8-2018q4/gcc-arm-none-eabi-8-2018-q4-major-linux.tar.bz2
RUN mkdir -p /compiler
RUN tar xfj gcc-arm-none-eabi-8-2018-q4-major-linux.tar.bz2 -C /compiler
ENV PATH /compiler/gcc-arm-none-eabi-8-2018-q4-major/bin:$PATH

RUN cd NanoVNA-H && make && ls

What are we missing in setting up the environment?

hugen79, 2021-12-11:
See this
https://github.com/DiSlord/NanoVNA-D/issues/1#issuecomment-698945620

closed 2021-12-11 by hugen79

jovo3, 2022-01-17:
That fixed it for me, too bad it can't build out of the box. For future reference, I did this:

- sed -i 's/hal_lld_backup_domain_init();/\/\/hal_lld_backup_domain_init();/g' NanoVNA-H/ChibiOS/os/hal/ports/STM32/STM32F0xx/hal_lld.c

- sed -i 's/hal_lld_backup_domain_init();/\/\/hal_lld_backup_domain_init();/g' NanoVNA-H/ChibiOS/os/hal/ports/STM32/STM32F3xx/hal_lld.c

- sed -i 's/OSAL_IRQ_HANDLER(STM32_ADC1_HANDLER) {/__attribute__((weak)) OSAL_IRQ_HANDLER(STM32_ADC1_HANDLER) {/g' NanoVNA-H/ChibiOS/os/hal/ports/STM32/LLD>

- and strip out the function in ChibiOS/os/hal/ports/STM32/STM32F3xx/hal_lld.c:54:
 static void hal_lld_backup_domain_init(void) {