hugen79/NanoVNA-H #71: Black screen on boot and failure to save anything to SD with 1.2.14 code

mkbodanu4, 2022-10-01:
I tried version 1.2.14 after 1.1.01 and found that 2 of 5 times the device won't boot at all, sticking on a black screen.
After turning it off and on again, it booted and worked well. Clearing configuration has not helped at all.

Also, I found that I was not able to create a screenshot on the SD card I used with the previous firmware. So I reverted back and all worked back again.

Then I tried DiSlord version 1.2.00 and no such issues occurred.

Device: NanoVNA-H, hardware version 3.6

pwschattenberg, 2022-12-10:
I have this same problem too. I am using the same device and hardware version as you.

Sarah-C, 2023-05-08:
Yup - same here, black screen on boot.