hugen79/NanoVNA-H #77: NanoVNA-H v 1.2.14 - SWR scan freezes for a second or two every so often.

Sarah-C, 2023-05-08:
Using V 1.2.14, I've got one trace running, SWR - giving a good trace of an antenna.

Every few seconds it pauses..... though the menu's still respond. I've not clicked the "Pause" button on the menu system.
Nothing else to report about reproducing the issue.

Post here if anyone else has noticed this - or if it's just me!

sergey12malyshev, 2023-08-28:
A similar problem. When you move the slider, the screen freezes. MS5351 , version 1.2.14. Has the author abandoned this project?

sergey12malyshev, 2023-08-31:
I found out that freezes occur when the serial port is turned on. To begin with, it's worth checking that CONFIG -> CONNECTION costs USB. It helps in part. Next, you can build the firmware with the __USE_SERIAL_CONSOLE__ macro disabled. I attach a link to my fork of the project with firmware: https://github.com/sergey12malyshev/NanoVNA-H/releases/tag/1.2.14%2B
https://github.com/sergey12malyshev/NanoVNA-H/tree/NoShell

I had freezes when working with the control wheel and just when working.

koriwi, 2023-09-19:
@sergey12malyshev thanks for your invenstigation! if it annoys me too much i know how to fix it. But for now i will leave it enabled as from time to time i use the serial connection :)