hugen79/NanoVNA-H #85: What are Format S11, Format S21, Channel S11

Sarah-C, 2024-08-11:
Is there a doc anywhere about the three "format, format, channel" buttons anywhere?

They're in the 1.2.27 build, but I'm a bit confused about how they flip, and what they're doing.

[image]

Sarah-C, 2024-08-12:
Nothing yet... I'll keep monitoring!

Sarah-C, 2024-08-12:
I noticed in ui.c, that the "channel" button mostly appears to toggle the kind of smith chart it uses - roughly speaking.

Any clearer explanation would be awesome! =)

`#define F_S11     0x00
#define F_S21     0x80
static UI_FUNCTION_ADV_CALLBACK(menu_format_acb)
{
  if (current_trace == TRACE_INVALID) return; // Not apply any for invalid traces
  uint16_t format = data & (~F_S21);
  uint16_t channel = data & F_S21 ? 1 : 0;
  if (b) {
    if (trace[current_trace].type == format && trace[current_trace].channel == channel)
      b->icon = BUTTON_ICON_CHECK;
    if (format == TRC_SMITH) {
      uint8_t marker_smith_format = get_smith_format();
      if ((channel == 0 && !S11_SMITH_VALUE(marker_smith_format)) ||
          (channel == 1 && !S21_SMITH_VALUE(marker_smith_format))) return;
      plot_printf(b->label, sizeof(b->label), "%s\n" R_LINK_COLOR "%s", get_trace_typename(TRC_SMITH, marker_smith_format), get_smith_format_names(marker_smith_format));
    }
    else
      b->p1.text = get_trace_typename(format, -1);
    return;
  }

  if (format == TRC_SMITH && trace[current_trace].type == TRC_SMITH && trace[current_trace].channel == channel)
    menu_push_submenu(channel == 0 ? menu_marker_s11smith : menu_marker_s21smith);
  else
    set_trace_type(current_trace, format, channel);
}

static UI_FUNCTION_ADV_CALLBACK(menu_channel_acb)
{
  (void)data;
  if (current_trace == TRACE_INVALID) {if (b) b->p1.text = ""; return;}
  int ch = trace[current_trace].channel;
  if (b){
    b->p1.text = ch == 0 ? "S11 (REFL)" : "S21 (THRU)";
    return;
  }
  set_trace_channel(current_trace, ch^1);
}
`

KDMcMullan, 2025-09-01:
I took some time to get my head around this, too. There may not be any nanoVNA specific info, since these channels are generic things that someone who uses a VNA would know. This was my first VNA, so I did not know!
The channel button allows you to select between S11 and S21 modes. S11 is "reflected" mode where you only use the S11 port (AKA port 1), which sends out a signal and watches for a reflected signal bouncing back. S21 is "through" mode where you use both ports: port 1 sends out a frequency and port 2 watches for the return signal.
Format S11 allows you to select the type of measurement to perform in S11 mode. Format S21 allows you to select the type of measurement to perform in S21 mode.