hugen79/NanoVNA-H #86: NanoVNA-H

TimBastress, 2024-08-14:
I'm trying to update the firmware in my NanoVNA-H to the latest shown here of v1.2.29 using the .dfu file and stm32CubeProgrammer software on Windows.

It seems to recongnize the connection and it reads it ok, and it seems to wirte the .dfu file ok as well but when I power off my vna and power it back on I just see a white screen. I cant seem to get it to boot back into the VNA screen.

[image]

any help would be greatly appreciated!

Tim

TimBastress, 2024-08-17:
the HW version on the back says 3.6_MS

It had a graphical video before i tried to upgrade the firmware. So I'm really confused as to why I cant get it to restart after the upgrade "seemingly successful"

TimBastress, 2024-08-20:
So apparently I've bricked my VNA while trying to do a firmware update since no one is responding back to me with a solution to my problem.  
Everything seemed to go well, but all I get is a white screen afterwards.  Even if I power down and back on still get white screen.  Surely someone out there has had this issue or knows what the issue is and can help me.  

What am I doing wrong?

ChrisHVL, 2024-08-20:
Hi,
maybe you should look here:

https://groups.io/g/nanovna-users/topic/nanovna_white_screen_usb_dead/71530182

Also there are many HAMs, which are using NANOVNA and have had similar experiences.
In my case, update via USB and DFUse had always worked.
vy73 DD6CN

TimBastress, 2024-08-21:
Thank you for the link above.   I'm rereading it again to make sure I'm not missing anything but I'm  still having issues. So it seems that when I try using dfuse demo software that it doesnt see the nanovna even though my windows computer sees it and the bootloader drivers activate and everything shows good the demo software doesnt seem to see it.    But when I use the STM32 Cube software it sees it just fine.   But neither are making a difference in getting my VNA to boot up properly.  This is very frustrating.   Normally updates go well for me via windows apps but not this one.

ChrisHVL, 2024-08-21:
Hi,
you may also look here:

https://nanovna.com/?page_id=103

There are several sites regarding NANOVNA...
In groups.io you may get support or in your local radio club. 
vy73