hugen79/NanoVNA-H #87: Screen garbage on first power up

etkal, 2024-10-02:
Just bought a new NanoVNA, HW 3.6.1_SM_ST, looks like the latest firmware from here (shows 1.2.27, and I even reloaded the latest image from here so it's identical).  When I first power it on after being off for a while, the screen flashes white, but then shows garbage on the upper half, which when written over by the display image still shows garbage data on the left side.  Powering off and back on fixes the issue, just flashes white, then display off, then draws as expected.

I suspect a timing issue on startup, might experiment with that, perhaps a delay is needed to allow the ili9341 HW to stabilize when cold.  I have a couple Raspberry Pi Pico projects that use the ili9341 and I had to play with the reset timing at startup to get it to behave properly:

void ILI934X::Reset()
{
    gpio_put(m_rst, 1);
    sleep_ms(50);
    gpio_put(m_rst, 0);
    sleep_ms(50);
    gpio_put(m_rst, 1);
    sleep_ms(50);
}

Have you seen this or had it reported before?  I'll see if I can experiment with this code, have not developed for this processor / SDK before.

  Regards,
  Erik

[image]

[image]

etkal, 2024-11-05:
Is anyone else seeing this??

hugen79, 2025-02-20:
The new version already performs lcd_clear_screen() once before entering the main boundary;

closed 2025-02-20 by hugen79